bosch-buderus-wp/bosch-buderus-wp.github.io #1: Genial

jh537, 2025-01-05:
Hi,

great idea and work to bundle all the information. The information has helped me a lot - especially the explanations and the dash as well as the sensors for HA. 

What else I would like to see:

- Change log on the website
- More detailed instructions for adjusting the primary-secondary circuits (spread/differential pressure setting)
    - Perhaps also a display in the HA
    - Auto optimization 
- Entity for the seasonal performance factor (incomplete in the Dash)

Many thanks for the effort....

Hi,

tolle Idee und Arbeit, die ganzen Infos mal zu bündeln. Mir haben die Infos ein großes Stück weiter geholfen-besonders die Erklärungen und das Dash sowie die Sensoren für HA. 

Was ich mir noch wünschen würde:

- Change-log auf der Webseite
- Detailliertere Anleitung zur Angleichung der Primär-Sekundärkreise (Spreizung/Differenzdruckeinstellung)
    - vielleicht auch eine Anzeige im HA
    - Autooptimierung 
- Entität zur Jahresarbeitszahl (unvollständig im Dash)

Vielen Dank für die Mühe....

jh537, 2025-01-05:
Hey, das ging ja Flott mit den Änderungen. Vielen Dank.

Kannst Du noch im Dash beim Balkendiagramm die AZ einfügen? Im neuen Dash fehlt noch der Sensor für die TAZ. Mega  Danke

jh537, 2025-01-05:
Idee für das Balkendiagramm:
Einbestellbar: AZ, VLT, AT, WWT
Danke

jh537, 2025-01-05:
- Erklärung AZ bei bestimmtem Verbrauch  kann besser/schlechter sein
- Relevanter ist aus meiner Sicht Verhältnis AT/IT/VT zu Verbrauch-->ggf. Darstellung?
-

jh537, 2025-01-07:
Wahrscheinlich sind die 250 mBar das Optimum für niedrige und höhere VL-Temp. Bei mir sind bei 250 eine Differenz von 1 Grad (T0-TC1). Damit habe ich die besten Werte grad. Ich hatte die 250 mBar nie verwendet, da das Pumpengeräusch nebenan zu hören war.

ASaeckl, 2025-01-14:
Hi,

ich darf mich dem Lob anschließen und finde die Zusammenfassung super! Die Einstellungen sind sehr gut erklärt und die ersten habe ich bereits umgesetzt! 

Eine Frage: Bei mir sind ein paar Sensoren nicht vorhanden und können im Dashboard nicht angezeigt werden. Wurden hierfür Helper erstellt oder liefert dir der ems-esp die Werte direkt? Folgende Sensoren sind nicht vorhanden:

- sensor.boiler_meterheat
- sensor.boiler_nrgheat
- sensor.boiler_wwmeter
- sensor.boiler_nrgww
- sensor.boiler_nrgtotal
- sensor.boiler_metercomp

Danke

mroeckl, 2025-01-15:
Vielen Dank, @ASaeckl! Ich habe dein Problem mit den fehlenden Entitäten in ein eigenes Issue verschoben: https://github.com/bosch-buderus-wp/bosch-buderus-wp.github.io/issues/2

bosch-buderus-wp, 2025-01-15:
Vielen Dank, @ASaeckl! Ich habe dein Problem mit den fehlenden Entitäten in ein eigenes Issue verschoben: https://github.com/bosch-buderus-wp/bosch-buderus-wp.github.io/issues/2

bosch-buderus-wp, 2025-01-15:
> Idee für das Balkendiagramm: Einstellbar: AZ, VLT, AT, WWT als Punkt/Linie

@jh537 , stellst du dir das in etwas so vor wie diese Dashboard Card: 

[image]

VLT, AT, WWT und Modulation kann man mit Klick auf die Legende einblenden.

Konfiguration findest du hier: https://github.com/bosch-buderus-wp/home-assistant/blob/main/dashboards/power-cop_card.yaml

jh537, 2025-04-27:
> [@jh537](https://github.com/jh537) , stellst du dir das in etwas so vor wie diese Dashboard Card:
>
> [image]
>
> VLT, AT, WWT und Modulation kann man mit Klick auf die Legende einblenden.
>
> Konfiguration [...]

Jo, das sieht prima aus-jetzt wäre noch eine Anzeige der TAZ gut als einzelne Kennzahl im Dash und die Angleichung der Volumenströme

better-tester, 2025-06-21:
Entities/Entitäten: **heatcable / Heizband**: Leitet man das Kondenswasser z.B. in die Kanalisation ein, gibt es das Heizband als Zubehör, um zu verhindern, dass der Ablauf bei tiefen Temperaturen zufriert. ems-esp definiert dies als Schalter, wird aber durch die WP in Abhängigkeit der Außentemperatur gesteuert. In HA ist das nur für Neugierige, um zu sehen, ob es denn auch funktioniert.

Ich habe folgende Erfahrung (CS6800i) gemacht (muss aber nicht allgemeingültig sein) und bei einem Update den Techniker ausgefragt. Kernsteuerung sind zwei Einheiten für die Inneneinheit und Außeneinheit (**boiler**) - dort sind die Einstellungen der WP hinterlegt. Für den "normalen" Benutzer ist das Display (Mäusekino) (**thermostat**) der Inneneinheit - also der Teil, den man verstellen darf, um zwar in der kalten Stube zu sitzen, aber ohne Schäden an der Technik anzurichten. Wenn es die Möglichkeit gibt, Dinge am **boiler** oder am **thermostat** zu verstellen, hat der **thermostat** das Sagen. Verstellt man den **boiler** werden die Änderungen durch die Einstellungen des  **thermostats** (teilweise) zurückgesetzt. Vielleicht hilft es euch bei euren Basteleien (Automatisierungen).

Danke für diese Sammelstelle zur Bosch/Buderus-Technik!