bosch-buderus-wp/bosch-buderus-wp.github.io #2: Fehlende Sensoren/Entitäten

bosch-buderus-wp, 2025-01-14:
Hi,

ich darf mich dem Lob anschließen und finde die Zusammenfassung super! Die Einstellungen sind sehr gut erklärt und die ersten habe ich bereits umgesetzt! 

Eine Frage: Bei mir sind ein paar Sensoren nicht vorhanden und können im Dashboard nicht angezeigt werden. Wurden hierfür Helper erstellt oder liefert dir der ems-esp die Werte direkt? Folgende Sensoren sind nicht vorhanden:

- sensor.boiler_meterheat
- sensor.boiler_nrgheat
- sensor.boiler_wwmeter
- sensor.boiler_nrgww
- sensor.boiler_nrgtotal
- sensor.boiler_metercomp

Danke

_Originally posted by @ASaeckl in https://github.com/bosch-buderus-wp/bosch-buderus-wp.github.io/issues/1#issuecomment-2589835987_

bosch-buderus-wp, 2025-01-17:
@ASaeckl, diese Entitäten kamen bei mir direkt durch ems-esp. Helper-Entitäten habe ich dafür nicht anlegen müssen.

Welche Wärmepumpe hast du? Und welche Version von ems-esp läuft bei dir?

bosch-buderus-wp, 2025-02-02:
Danke, @Fuzzy923 , für die Auflistung.
Laut [Entity Register](https://docs.emsesp.org/Modbus-Entity-Registers/#enviline-compress-6000aw-hybrid-3000-7000iaw-supraecogeo-5xx-wlw196iwsw196i) von ems-esp müssten die fehlenden Entitäten wie `nrgtotal` oder `nrgheat` auch für die WLW196i verfügbar sein.
In deiner Liste stehen die _full names_ der Entitäten.  In neueren Versionen von ems-esp wird aber der _short name_ in der MQTT Discovery verwendet. Daher wäre es nett, wenn du dir diese _short names_ mal anschauen könntest. Die  Liste aller Boiler-Entitäten mit _short name_ kannst du einfach unter folgender Adresse aufrufen: [http://ems-esp/api/boiler/](http://ems-esp/api/boiler/)
Prüf doch mal, ob die fehlenden Entitäten in dieser Liste enthalten sind.

roydigerhund, 2025-03-12:
Die Werte nrgtotal, nrgheat, nrgcool und dhw/nrg sind bei der WLW196i leer, aber die Werte für nrgcons* sind vorhanden.

[image]

bosch-buderus-wp, 2025-03-15:
Danke, @roydigerhund , für die Analyse.
Ich habe kürzlich versucht, die ganzen Entitäten für mich etwas zu strukturieren. Meine Erkenntnisse habe ich hier notiert: [https://bosch-buderus-wp.github.io/docs/smarthome/entities](https://bosch-buderus-wp.github.io/docs/smarthome/entities).
Bei meiner Bosch CS6800i ist `nrgtotal=nrgsupptotal`, wobei `nrgtotal` mit 2 Nachkommastellen ausgegeben wird und `nrgsupptotal` ohne Nachkommastellen:

| 2 Nachkommastellen | Keine Nachkommastelle |
| --------------------- | ------------------------ |
| nrgtotal | nrgsupptotal |
| nrg | nrgsupp |
| nrgheat | nrgsuppheating |
| nrgcool | nrgsuppcooling |

@roydigerhund , könntest du bitte noch prüfen, ob deine WLW 196i Werte für Entitäten, die mit `meter*`beginnen, liefert?
Bei meiner Bosch CS6800i sehe ich folgendes:

[image]

roydigerhund, 2025-03-18:
@bosch-buderus-wp die `meter*` Werte sind bei der WLW 196i auch alle leer.

Ich hab die aktuellen Werte exportiert und als CSV angehängt, wenn ich irgendwie helfen kann beim bestimmen von Werten, melde dich gerne.

[Kessel_WP_CUH_Enviline, Compress 6000AW, Hybrid 3000-7000iAW, SupraEco_Geo 5xx, WLW196i_WSW196i-2.csv](https://github.com/user-attachments/files/19329558/Kessel_WP_CUH_Enviline.Compress.6000AW.Hybrid.3000-7000iAW.SupraEco_Geo.5xx.WLW196i_WSW196i-2.csv)

[Thermostat_HMC310_RC3*0, Moduline 3000_1010H, CW400, Sense II, HPC410-2.csv](https://github.com/user-attachments/files/19329560/Thermostat_HMC310_RC3.0.Moduline.3000_1010H.CW400.Sense.II.HPC410-2.csv)

mtothearkus, 2025-03-22:
Hi,

ich habe eine Buderus WLW186i und das BBQkeys Gateway angeschlossen.
Die nrg-Entitäten habe ich im ems-esp, unter HA aber irgendwie nicht. Dementsprechend sieht das Dashboard entsprechend leer aus.

[image]

[image]

mtothearkus, 2025-03-22:
ok, ich habe das problem glaube ich selber gelöst bekommen:
im ems-esp musste ich unter mqtt eine Kleinigkeit ändern:

[image]
ganz hinten von von Einzelinstanz, Langname 3,4 auf => Einzelinstanz, MQTT Name 3,6

Fuzzy923, 2025-03-25:
Ich habe die gleichen Einstellungen im EMS-ESP und deshalb das Dashboard auf die Entitäten umgebogen und optimiert:
[image]

Fuzzy923, 2025-03-25:
Ich habe einen Wasserspeicher mit zwei Sensoren. Der obere ist `sensor.boiler_wwcurtemp` und der untere `sensor.boiler_hptw1`. Meine Beobachtungen haben gezeigt, dass beim Warmwasser nicht `sensor.boiler_wwcurtemp` als Ausschaltsignal herangezogen wird, bis `sensor.boiler_wwsettemp` erreicht wird, sondern `ensor.boiler_hptc3`

roydigerhund, 2025-03-26:
@Fuzzy923 die beiden Sensoren `sensor.boiler_wwcurtemp2` und `sensor.boiler_hptw1` liefern bei mir die gleichen Werte, `sensor.boiler_wwcurtemp` hab ich als Temperatur für die Wärmetauscher-Spirale im Speicher vermutet, da diese nicht der Wassertemperatur entspricht die aus dem Hahn kommt, vor allem nicht nach längerer Warmwassernutzung. Das ist allerdings der Wert der auf dem Display der Wärmepumpe als WW-Temperatur benutzt wird.

Das Warmwasser-Programm schaltet sich bei mir ebenfalls ab, wenn `sensor.boiler_hptc3` die Soll-Temperatur erreicht. Ich verstehe allerdings die Logik dahinter nicht, ich bin davon ausgegangen, dass TC3 ein Sensor im Kältekreis ist.

[image]

bosch-buderus-wp, 2025-03-26:
> Langname 3,4 auf => Einzelinstanz, MQTT Name 3,6

Danke für den Hinweis @mtothearkus ! Ich habe die Info in der [Projektbeschreibung](https://github.com/bosch-buderus-wp/home-assistant/commit/2a6eb6fd96fa368389d289293924960adfd51b80) aufgenommen.

roydigerhund, 2025-04-09:
@bosch-buderus-wp hast du eine Idee wie ich herausfinde, wo genau die Sensoren im Warmwasserspeicher die Werte messen?

Xanti19, 2025-05-18:
> Ich habe die gleichen Einstellungen im EMS-ESP und deshalb das Dashboard auf die Entitäten umgebogen und optimiert: [image]

Moin, kannst Du bitte den YAML-Code für Dein Dashboard hier teilen? Ich kämpfe aktuell noch mit den vielen unterschiedlichen Bezeichnungen der Entitäten.

Xanti19, 2025-05-19:
@bosch-buderus-wp   
Zunächst ein großes Lob von mir für Deine hier bereitgestellten Infos. Seit Freitag ist bei uns eine Buderus WLW186i-12 E mit dem EMS-ESP Gateway in Betrieb.
Deine Entitäten-Übersicht mit den Erklärungen ist bei der Masse an Entitäten, die vom Gateway zur Verfügung gestellt werden, sehr hilfreich.  
Bei dem hier vorgestellten einfachen Dashboard kann ich allerdings zwei Entitäten nicht zuordnen. Bei den aktuellen Werten wird bei der elektrischen Leistungsaufnahme 'sensor.boiler_powerconstotal' und bei der thermischen Leistungabgabe 'sensor.boiler_powertotal' angegeben. Hier sollen die aktuellen Werte ausgeben werden,  nrgtotal und metertotal würden dann nicht passen, oder?

thoughtgap, 2025-05-19:
@Xanti19 lies mal hier: https://bosch-buderus-wp.github.io/docs/smarthome/ha#arbeitszahlcop-mit-helfer-entit%C3%A4ten

Xanti19, 2025-05-20:
Die Automatisierung 'force_update' funktioniert bei mir nicht. Liegt wahrscheinlich an der Pfadangabe im Topic. Habe mir nun zwei Helfer jeweils als Verbrauchszähler angelegt; einmal 'Verbrauch WP Tag' mit der Entität _sensor.boiler_meter_total_ und 'Energieabgabe WP Tag' mit der Entität _sensor.boiler_total_energy_supplied_. Diese Helfer verwende ich nun als Eingangssensor für _boiler_powertotal_ und _boiler_powerconstotal_.

[image]

Die Ausgabe der Vorlauf- und Rücklauftemperatur (_sensor.boiler_current_flow_temperature_ + _sensor.boiler_return_temperature_)  ist eher missverständlich. Am BC400 werden als Vorlauftemperatur 37° und als Rücklauftemperatur 32° angezeigt. Für die "richtige" Vorlauftemperatur habe ich noch keine Entität gefunden.

bosch-buderus-wp, 2025-05-25:
> [@bosch-buderus-wp](https://github.com/bosch-buderus-wp) die `meter*` Werte sind bei der WLW 196i auch alle leer.

Danke @roydigerhund für die CSV-Dateien und entschuldige die späte Antwort.
Ich habe die Liste der [Smarthome Entitäten](https://bosch-buderus-wp.github.io/docs/smarthome/entities) mit deiner CSV verglichen und folgende Entitäten scheinen bei der WLW 196i nicht verfügbar zu sein:
- Energiewerte mit 2 Nachkommastellen, d.h. die Entitäten, die mit `nrg...` oder `meter...` beginnen
- Ladedelta- und Stopptemperaturen, d.h. die Entitäten, die mit `...diff` oder `...stop` enden - die _Stopptemperatur_ heißt bei der WLW 196i vermutlich _Ausschalttemp._, _Ladedelta_ konnte ich bisher nicht finden
- Entitäten zu PC0 oder PC1
- Systemdruck `syspress`
- Anzahl der Kompressorstarts für Warmwasserbetrieb `starts2` - könnte sich bei der WLW 196i hinter _WWK Anzahl Starts WP_ verbergen

Ich hab die Verfügbarkeit in folgendem Commit [58a1021](https://github.com/bosch-buderus-wp/bosch-buderus-wp.github.io/commit/58a1021376af9b33d840cc48f79ee838eabdaf35) eingetragen, hätte aber noch 2 Fragen:
1. Die Ladedeltaeinstellungen für die Warmwasseraufbereitung konnte ich in der CSV-Datei nicht finden. Kann man das bei der WLW 196i nicht einstellen?
2. Könntest du mir für die WLW 196i noch kurz schreiben, wie die ID zu _WWK Anzahl Starts WP_ lautet? Ich meine damit, unter welchem Property-Name die Entität unter [http://ems-esp/api/boiler/entities](http://ems-esp/api/boiler/entities) auftaucht. Dann trage ich das noch auf [Smarthome Entitäten](https://bosch-buderus-wp.github.io/docs/smarthome/entities) nach.

roydigerhund, 2025-05-28:
Wo würde ich die Ladedeltaeinstellungen finden? Ich hab keine unter den einzelnen Einstellungen für die Warmwasser-Programme, sondern nur eine allgemeine Temperatur-Differenz Einstellung für TC3/TC0 für Heizen und kühlen.

[image]

[image]

Die Daten der Entities für die Starts sind wie folgt:

```json
{
  "totalcompstarts": {
    "name": "totalcompstarts",
    "fullname": "Gesamtkompressorstarts ",
    "circuit": "",
    "value": 4747,
    "type": "number",
    "readable": true,
    "writeable": false,
    "visible": true
  },
  "heatingstarts": {
    "name": "heatingstarts",
    "fullname": "Heizungsregelungstarts",
    "circuit": "",
    "value": 3084,
    "type": "number",
    "readable": true,
    "writeable": false,
    "visible": true
  },
  "coolingstarts": {
    "name": "coolingstarts",
    "fullname": "Kühlregelungstarts",
    "circuit": "",
    "value": 3,
    "type": "number",
    "readable": true,
    "writeable": false,
    "visible": true
  },
  "dhw.startshp": {
    "name": "startshp",
    "fullname": "WWK Anzahl Starts WP",
    "circuit": "dhw",
    "value": 1660,
    "type": "number",
    "readable": true,
    "writeable": false,
    "visible": true
  }
}
```

Das stimmt auch mit den Daten überein, die in der WP zu sehen sind:

[image]

bosch-buderus-wp, 2025-06-09:
Vielen lieben Dank, @roydigerhund, für die ausführliche Antwort.

**Ladedelta**: Ladedelta (aka: Temperaturdifferenz für Beladung) kann man bei der Bosch CS5800/6800i und Buderus WLW176/186 unter `Anlageneinstellungen` &rarr; `Warmwasser` einstellen. Aber nachdem ich es in deinen CSV-Dateien nicht gesehen habe, vermute ich, dass die WLW 196i diese Einstellung nicht unterstützt.

**Starts WW**: Danke, ich hab die Entität `startshp` auf [Smarthome Entitäten](https://bosch-buderus-wp.github.io/docs/smarthome/entities#kompressorstarts) nachgetragen und vermerkt, dass es diese nur bei der WLW 196i gibt.

Mit den Erkenntnissen aus diesem Issue habe ich auch ein angepasstes Home Assistant Dashboard mit den Entitäten der WLW196i hinzugefügt: https://github.com/bosch-buderus-wp/home-assistant/commit/3a0a64afe9ed13108f36c13dcbd9ff11b9754d16

Damit schließe ich dieses Issue, in dem es um fehlende Sensoren/Entitäten der WLW196i ging. Die offene Diskussion zu den Sensoren/Entitäten des Warmwassers und auch die Frage von @Xanti19 starte ich im [Diskussionsforum](https://github.com/bosch-buderus-wp/bosch-buderus-wp.github.io/discussions).

closed 2025-06-09 by bosch-buderus-wp