bosch-buderus-wp/bosch-buderus-wp.github.io #3: WLW196i ?

Deepintheeast, 2025-01-16:
Hi, tolles Projekt!

Gibt es eventuell schon Ansätze das ganze auch für die WLW196i umzusetzen? 
Leider sehen die Entitäten da komplett anders aus!?
Muss mich da wohl, wenn es die Zeit hergibt, mal etwas intensiver einarbeiten!

Wie schon gesagt, Tolles Projekt! -> mache weiter so!

Grüße....

roydigerhund, 2025-03-11:
Ich hab auch die WLW196i und bin gerade dabei herauszufinden welche Werte ich mir am besten anschauen sollte um die Nutzung der Wärmepumpe zu verbessern.   
  
Vieles aus der existierenden [Entitäten-Liste](https://bosch-buderus-wp.github.io/docs/smarthome/entities) hat bei mir gestimmt.

bosch-buderus-wp, 2025-06-09:
@Deepintheeast , mit Hilfe von @roydigerhund und @Fuzzy923 habe ich die Entitäten, die nur auf der CS5800/6800i & WLW176/186i beziehungsweise nur auf der CS7001i & WLW196i verfügbar sind, in der [Entitäten-Liste](https://bosch-buderus-wp.github.io/docs/smarthome/entities) markiert. Die wesentlichen Unterschiede sind:
- [Genaue Energiewerte](https://bosch-buderus-wp.github.io/docs/smarthome/entities#mit-2-nachkommastellen) sind nur bei der CS5800/6800i & WLW176/186i verfügbar
- [Ladedeltas der Warmwasseraufbereitung](https://bosch-buderus-wp.github.io/docs/smarthome/entities#einstellungen-1) sind nur bei der CS5800/6800i & WLW176/186i verfügbar und Stopptemperatur hat einen anderen Entitätsnamen
- [Entitäten der Pumpen PC0 und PC1](https://bosch-buderus-wp.github.io/docs/smarthome/entities#pumpen) sind nur bei der CS5800/6800i & WLW176/186i verfügbar
- [Wasserdruck im Heizkreis](https://bosch-buderus-wp.github.io/docs/smarthome/entities#status) ist nur bei der CS5800/6800i & WLW176/186i verfügbar
- Die Entität für [Anzahl der Kompressorstarts für Warmwasserbetrieb](https://bosch-buderus-wp.github.io/docs/smarthome/entities#kompressorstarts) heißt bei der WLW196i `startshp` anstatt `starts2`

Das sind zumindest die Unterschiede bei den wichtigsten Entitäten.

Das Beispiel-Dashboard für Home Assistant hab ich entsprechend angepasst: https://github.com/bosch-buderus-wp/home-assistant/blob/main/dashboards/simple-dashboard-CS7001i-WLW196i.yaml

Ich schließe damit dieses Issue.

closed 2025-06-09 by bosch-buderus-wp