bosch-buderus-wp/bosch-buderus-wp.github.io #4: PV Integration

CR4567, 2025-02-22:
Hi,
erstmal ein Kompliment und Dank für die großartige Idee, mal alle Informationen an einem Ort zu sammeln.
Wir sind seit September auch im Besitz der Bosch 6800 und ich bin seit dem am Foren durchforsten nach den passenden Einstellungen und Integrationen, speziell für HomeAssistant.
Tatsächlich habe ich es auch schon über den ESP integriert. 
Da wir auch im Besitz einer PV Anlage sind, kam nun auch die Idee auf, bei genügend PV Leistung mehr Warmwasser zu produzieren, etc.
Die Wärmepumpe unterstützt das ja theoretisch schon und besitzt dafür einen eigenen Eingang.

Hattest du dich mit dem Thema schon auseinandergesetzt oder Informationen dazu?

Tihmann, 2025-02-23:
Hier gibt es ein paar Informationen https://bbqkees-electronics.nl/2024/10/03/using-the-smart-grid-sg-and-photovoltaic-pv-function-of-your-heat-pump-with-the-ems-gateways/

CR4567, 2025-02-27:
ah danke für den Tip, damit könnte es tatsächlich klappen. Ich werde es am Wochenende testen und berichten.

CR4567, 2025-03-09:
Noch nicht so viel wie ich wollte. 
Sagen kann ich aber schon mal, dass es tatsächlich funktioniert über das ems gateway den Wert zugehörigen Wert für PV oder SG einfach auf 1 zu setzen. Die Wärmepumpe zeigt dann im Display auch an, dass PV Strom aktiv ist.
Allerdings habe ich noch 3 Themen die ich mir genauer ansehen muss:
- Welche Werte machen in der WP Sinn bei aktivem PV Strom?
- Welcher Eingang lohnt sich eher für HA, der PV oder SG Eingang?
- Logik in HA wann Eingang auf aktiv wechselt und wann zurück

Sobald ich dazu mehr habe, gebe ich die Erfahrungen hier gerne weiter.

Tihmann, 2025-03-25:
schau mal bei evcc https://evcc.io/ Da sind jetzt Wärmepumpen integriert, auch Buderus mit emsesp

bosch-buderus-wp, 2025-04-05:
Hallo @Tihmann , vielen Dank für die Integration in evcc. Hab es heute getestet. Ich kenne mich aber noch nicht so gut mit evcc aus und bin daher nicht sicher, ob ich alles richtig gemacht habe. Daher hätte ich noch ein paar Fragen:

- Wenn ich in evcc die Wärmepumpe auf `PV` stelle, passiert gar nichts. Nur wenn ich `Min+PV` oder `Schnell` auswähle, wechselt `hpin4opt` auf `100000100000`.
- Ich habe im Service-Menü der Wärmepumpe unter `Anlageneinstellungen` &rarr; `Photovoltoikanlage` die `Erhöhung der Wunschtemp. beim Heizen` auf 2K eingestellt und `Erhöhter Warmwasserkomfort` eingeschaltet. Ersteres scheint zu funktionieren, aber bei `Erhöhter Warmwasserkomfort` sehe ich keinen Effekt. Hätte erwartet, dass "WWK gewählte Temperatur" (dhw.seltemp) bei PV-Überschuss auf die Komfort-Stopptemperatur erhöht wird.

Funktioniert das bei dir?

bosch-buderus-wp, 2025-04-05:
Der zweite Punkt scheint doch wie erwartet zu funktionieren. Hab die falsche Entität angeschaut. "WWK Solltemperatur" ist jetzt auf der  Komfort-Stopptemperatur.

thoughtgap, 2025-04-06:
Mir fehlt im Service Menü unter `Anlageneinstellungen` der Menüpunkt `Photovoltaikanlage`. Weißt du zufällig wo man den herzaubern kann? Unter `Anlageneinstellungen` → `Manuelle Inbetriebnahme` könnte man nur `Solar` oder `Energiemanager` aktivieren.
Compress CS6800iAW 12M Inneneinheit hier, mit Software 9.7.0 und 47.11-Field beim Bedienfeld.

Tihmann, 2025-04-06:
Unter Anlageneinstellungen / Externer Eingang 4 / PV

bosch-buderus-wp, 2025-04-06:
Hab angefangen, die Integration der Wärmepumpe in evcc zu beschreiben. Siehe https://bosch-buderus-wp.github.io/docs/smarthome/evcc. Falls ich was falsch verstanden habe, gerne melden.

thoughtgap, 2025-04-13:
Ich setze die PV-Überschuss-Einstellung seit ein paar Tagen erfolgreich mit folgenden Automatisierungen direkt aus Home Assistant heraus. Ggf abweichende Sensor-Namen beachten:

```yaml
sequence:
  - target:
      entity_id: text.boiler_input_4_options
    data:
      value: >
        {% set current = states('text.boiler_input_4_options') %} {% if current
        | length > 0 %} 1{{ current[1:] }} {% else %} 1 {% endif %}
    action: text.set_value
alias: "Wärmepumpe: PV Überschuss an"
description: ""
```

```yaml
sequence:
  - target:
      entity_id: text.boiler_input_4_options
    data:
      value: >
        {% set current = states('text.boiler_input_4_options') %} {% if current
        | length > 0 %} 0{{ current[1:] }} {% else %} 0 {% endif %}
    action: text.set_value
alias: "Wärmepumpe: PV Überschuss aus"
description: ""
```

better-tester, 2025-06-05:
Kleine PV-Anlage und Bosch WP. EMS-ESP, weil zu faul immer in den Keller zu laufen und ein wenig HA. Ziel war vom Überschuss möglichst wenig dem EVU überlassen und bei HA **keep it simple** zu verfolgen, solange kein E-Auto auf dem Hof steht und evcc nicht zwingend erforderlich ist. Bei zwei Personen ist der Wasserverbrauch überschaubar und WW macht die WP in der Regel einmal am Tag. Also muss man die WP dazu überreden, auch mal WW zu machen, wenn die min-Temp des eingestellten Modus noch nicht erreicht ist.
Ich habe die Modi für das WW umdefiniert: ECO (40 - 48), Komfort (45 - 51). Dazu die Automatisierung mit folgender Logik:
Trigger(1) 5 min nach voller Stunde, wenn die Zeit zwischen 13:30 und 16:30 ist und gerade kein WW gemacht wird und die Temp unter 45° ist und min. 2000W vom Dach kommen, dann stelle den Modus auf Komfort. 

Trigger (2) Statuswechsel WP von WW auf irgendwas anderes und WW durch Trigger (1) initiiert wurde, dann setze den Modus auf den vorherigen Wert zurück.

Ein paar Zustände muss man sich merken, aber als Dünnbrettbohrer-Lösung funktioniert es, solange niemand an der Heizung rumspielt.  Beispiel gerne bei Bedarf.

markus5790, 2025-08-20:
Muss in evcc noch etwas spezielles konfiguriert werden, damit bei Überschuss Ext4 invertiert wird oder reicht es die Wärmepumpe über ems-esp mit evcc zu verbinden?

markus5790, 2025-08-20:
Genau nach dieser Einleitung habe ich es eingerichtet, allerdings war mir nicht klar, ob irgendwo noch weitere Konfigurationen für die invertierung gemacht werden müssen. Werde es morgen nochmal beobachten, wenn Warmwasser erzeugt werden soll.

markus5790, 2025-08-21:
Ich glaube es funktioniert nicht. Die WP macht Grade Warmwasser und es ist Übeschuss vorhanden. Dies wird auch von evcc erkannt und der 60s Timer fängt an zu laufen. Nach Ablauf des Timers erfolgt ein Error:

```
[site ] DEBUG 2025/08/21 14:20:44 site power: -2576W
[lp-1 ] DEBUG 2025/08/21 14:20:44 charger status: B
[lp-1 ] DEBUG 2025/08/21 14:20:44 charger soc limit: 46%
[lp-1 ] DEBUG 2025/08/21 14:20:44 pv charge current: 16.6A = 5.44A + 11.2A (-2576W @ 1p)
[lp-1 ] DEBUG 2025/08/21 14:20:44 site power -2576W <= 0W enable threshold
[lp-1 ] DEBUG 2025/08/21 14:20:44 pv enable in 0s
[lp-1 ] DEBUG 2025/08/21 14:20:44 pv enable timer elapsed
[lp-1 ] DEBUG 2025/08/21 14:20:44 pv timer reset
[lp-1 ] DEBUG 2025/08/21 14:20:44 pv timer inactive
[lp-1 ] ERROR 2025/08/21 14:20:44 charger enable: unexpected status: 401 (Unauthorized)
```

Ich habe in ems-esp einen neuen Admin Nutzer angelegt und den token in die evcc Konfiguration übertragen.

markus5790, 2025-08-21:
Es hat jetzt doch noch geklappt. Ich habe den Rechner auf dem HA läuft neu gestartet, danach hat die Kommunikation zwischen evcc und dem EMS geklappt.