bosch-buderus-wp/bosch-buderus-wp.github.io #5: Warmwasser Moduswahl

CR4567, 2025-02-27:
Hi,

ich würde meine Bosch Wärmepumpe gerne wie hier beschrieben https://bosch-buderus-wp.github.io/docs/einstellungen/#modusauswahl 
über Profile das WW heizen lassen. Ich kann diese Option aber nicht finden. Ist die von einer bestimmten Komponente abhängig?
Installiert ist eine Bosch 6800 iAW 12 mit UI 800. In der Anleitung konnte ich auch nichts dazu finden.

mroeckl, 2025-02-27:
Hi @CR4567, auf meiner Bosch 6880i AW 4 mit UI 800 kann ich die Zeitprogramme einstellen, wenn ich auf dem Startbildschirm auf _Warmwasser_ drücke und dann _Mehr...> → Zeitprogramm → Bearbeiten_ . Oder in der App ebenfalls auf dem Startbildschirm unter _Warmwasser_  und dann auf dem Kalendersymbol unten rechts.

CR4567, 2025-02-28:
Hi @mroeckl  und vielen Dank. Ich war blind vor lauer tiefergehender Einstellungen und bin gar nicht auf die Idee gekommen mal in der Startoberfläche nachzusehen.
Das hat mir gerade unglaublich geholfen. Hatte schon befürchtet das wird einfach nicht angeboten.

closed 2025-02-28 by CR4567