bosch-buderus-wp/bosch-buderus-wp.github.io #6: Home Assistant: Falsche Arbeitszahl bei Inaktivität der WP

thoughtgap, 2025-04-05:
Hallo und vielen Dank für diese gute Doku.
Ich habe eine Bosch AW 5 OR-S CS6800iAW mit Compress CS6800iAW 12M Inneneinheit. Die Namen einiger Sensoren weichen leicht von dieser Doku ab, über die Namen lassen sie sich jedoch gut mappen.

Ich habe die berechneten Sensoren in Home Assistant erstellt:
* Elektrische Leistungsaufnahme (10min Ableitungssensor auf Gesamtmessung `sensor.boiler_meter_total`)
* Thermische Leistungsabgabe (10min Ableitungssensor auf Gesamtenergie `sensor.boiler_total_energy`)
* Arbeitszahl (Thermische Leistungsabgabe/Elektrische Leistungsaufnahme)

## Problem
Bei längerer Inaktivität der Wärmepumpe wird die `Gesamtenergie` nicht mehr geupdated, da keine Energie abgegeben wird.
Die `Gesamtmessung` wird weiterhin geupdated, da der Leerlauf der Anlage ca. 30W hat.

Der Ableitungssensor `Thermische Leistungsabgabe` hängt dann auf dem letzten Wert fest, was eine unrealistische Arbeitszahl ergibt.

[image]

[image]

[image]

[image]

## Workaround
Um diese Problematik zu handhaben habe ich einen neuen Template-Sensor mit einer bereinigten Thermischen Abgabe erstellt, der `0` ausgibt falls das letzte Update der Thermischen Leistungsabgabe älter als 10 Minuten ist.

3. **Thermische Leistungsabgabe bereinigt** als _Template für einen Sensor_
   - Art: Helfer &rarr; Template &rarr; Template für einen Sensor
   - Name: _wp_thermische_leistungsabgabe_cleansed_
   - Zustandstemplate:
     {% raw %}
     ```
     {% set thermal_state = states('sensor.wp_thermische_leistungsabgabe') %}
     {% set last_updated = states.sensor.wp_thermische_leistungsabgabe.last_updated %}
     {% set current_time = now() %}
     {% set time_diff = (current_time - last_updated).total_seconds() | int %}

     {% if thermal_state != 'unavailable' and thermal_state != 'unknown' and time_diff < 600 %}
     {{ thermal_state | float }}
     {% else %}
     0
     {% endif %}
     ```
     {% endraw %}
   - Geräteklasse: _Leistung_
   - Zustandsklasse: _Messung_
   - Gerät: _ems-esp Boiler_

Die Arbeitszahl kann dann auf Basis des bereinigten Wertes berechnet werden:

```
{% set q = states('sensor.wp_thermische_leistungsabgabe_cleansed') | float %}
{% set p = states('sensor.boiler_powerconstotal') | float %}
{% if q > 0 and p > 0 %}
  {{ (q / p) | round(2) }}
{% else %}
  0
{% endif %}
```

[image]

Habt ihr das Phänomen "hängende Gesamtenergie" ebenfalls?

thoughtgap, 2025-04-06:
Problem mit meiner Korrektur: In den 10 Minuten zwischen Abschaltung der WP und Greifen des per Formel realisierten Timeouts ist auch die korrigierte thermische Leistungsabgabe inkorrekt. Der Wert des Quellsensors hängt in diesem Zeitraum noch fest auf dem letzten gültigen Wert. In der Leistungsabgabe werden somit die eigentlichen Nullwerte im 10min-Zeitraum nicht berücksichtigt.

mroeckl, 2025-04-11:
Danke @thoughtgap für deine Verbesserung.
Soweit ich weiß, ist die Ursache der "hängenden Gesamtenergie" ein seit langem bekanntes und oft diskutiertes Problem in Home Assistant, dass Ableitungssensoren bei unveränderten Werten "hängen" bleiben (siehe z.B. https://github.com/home-assistant/core/issues/83496).
Ich probiere gerade ein paar alternativen Lösungsideen, wie z.B. ein `force_update: true` auf der MQTT-Entität. Ich melde mich, sobald ich erste Ergebnisse habe.

mroeckl, 2025-04-12:
Kurzes Update: Ich habe gestern Abend `force_update: true` auf der MQTT-Entität `ems-esp Boiler Gesamtenergie (boiler_nrgtotal)` gesetzt. Das sollte dazu führen, dass jeder neue Wert zu einer Aktualisierung führt, auch wenn es keine Veränderung gab. Wie man im Screenshot unten sieht, führt das dazu, dass der Ableitungssensor `boiler_powertotal` wirklich 0 wird, wenn sich die Werte von `boiler_nrgtotal` nicht mehr ändern. Im Screenshot ist die WP zwischen 9-13 Uhr und ab 14:30 Uhr aus und `boiler_powertotal`ist in diesen Zeiträumen 0.

[image]

Auch im Dashboard wird die Thermische Leistungsabgabe mit dem Wert 0 richtig dargestellt und die Arbeitszahl ist auch 0:

[image]

Ich schau mir gerade noch an, wie man `force_update: true` einfach in die MQTT-Discovery für `boiler_nrgtotal` mit einer Automation integrieren kann.

@thoughtgap , was hältst du von der Lösung?

thoughtgap, 2025-04-13:
Dein Vorgehen gefällt mir. Wo (ESP/MQTT-Broker/Homeassistant) und wie hast du den `force_update` Parameter gesetzt? Danke

mroeckl, 2025-04-13:
Ich denke ein einfacher Weg, um `force_update` auf einer MQTT-Entität zu setzen und das MQTT-Autodiscovery beizubehalten, ist eine Automation  unter `Einstellungen` &rarr;  `Automationen & Szenen` &rarr; `+ AUTOMATION ERSTELLEN (rechts unten` &rarr; `Neue Automation erstellen` &rarr; `... (3 Punke rechts oben)` &rarr; `In YAML bearbeiten` und dann folgende Konfiguration einfügen:
```
alias: Set force_update on boiler_nrgtotal
description: "This automation adds 'force_update: true' to the discovery message of 'boiler_nrgtotal', causing the value to be updated regularly, even if it remains unchanged. This ensures that helper entities, such as derivatives, function reliably."
triggers:
  - trigger: time_pattern
    hours: /1
    minutes: "0"
    seconds: "0"
conditions: []
actions:
  - action: mqtt.publish
    metadata: {}
    data:
      retain: true
      topic: homeassistant/sensor/ems-esp/boiler_nrgtotal/config
      payload: |
        {
          "uniq_id": "boiler_nrgtotal",
          "obj_id": "boiler_nrgtotal",
          "name": "Gesamtenergie",
          "stat_t": "ems-esp/boiler_data",
          "val_tpl": "{{ "{{ value_json['nrgtotal'] if value_json['nrgtotal'] is defined else 0 }}" }}",
          "avty": [
            {
              "t": "ems-esp/boiler_data",
              "val_tpl": "{{ "{{ 'online' if value_json['nrgtotal'] is defined else 'offline' }}" }}"
            }
          ],
          "avty_mode": "all",
          "unit_of_meas": "kWh",
          "stat_cla": "total_increasing",
          "dev_cla": "energy",
          "dev": {
            "ids": [
              "ems-esp-boiler"
            ],
            "name": "ems-esp Boiler"
          },
          "force_update": true
        }
mode: single

```
Dann speichern und zur nächsten Stunde sollte dann die `boiler_powertotal` ordnungsgemäß funktionieren, wenn ich keinen Denkfehler habe. Würde mich sehr interessieren, ob es bei dir auch funktioniert. Dann könnte ich die Lösung auf der Webseite veröffentlichen.

thoughtgap, 2025-04-13:
Super, vielen Dank! Ich musste meine MQTT-Message leicht anpassen, denn meine Entitätsnamen weichen ab.
Das Beobachten von möglichen Veränderungen mit [MQTT Explorer](https://mqtt-explorer.com/) hat geholfen, damit nichts unnötig verändert wird.

Der Sensor heißt bei mir `boiler_total_energy` statt `boiler_nrgtotal`). Im EMS-ESP lautet meine Einstellung bei `Entitäts-ID Format` "Einzelinstanz, Langname (v3.4)

[image]

Hier würde mich deine Einstellung interessieren, die Wahl des Formats wäre auch für diese Doku interessant (denn alle yamls passen bei mir nicht 1:1 und müssen angepasst werden).

Die Automatisierung läuft nun und bislang sieht es gut aus. Ich beobachte das Verhalten und melde mich erneut!

jh537, 2025-04-13:
> 
> Dann speichern und zur nächsten Stunde sollte dann die `boiler_powertotal` ordnungsgemäß funktionieren, wenn ich keinen Denkfehler habe. Würde mich sehr interessieren, ob es bei dir auch funktioniert. Dann könnte ich die Lösung auf der Webseite veröffentlichen.

Hier hat es leider nicht mit dem Verfahren oben funktioniert. Ich habe das YAML in /homeassistant/automations.yaml eingefügt. Damit ist die AZ schon mal sofort auf 0 ordnungsgemäß gegangen. Ich habe eine Installation auf einem Raspberrypi-liegt sicher an verschienden Pfaden.

thoughtgap, 2025-04-15:
Update: Das setzen des `force_update` Flags hilft auch bei mir, die Home Assistant Entität wird weiterhin aktualisiert und der Ableitungssensor hat einen gültigen Wert.
Anstelle der stündlichen Aktualisierung probiere ich nun, Nachrichten ohne das Flag mit diesem zu ergänzen:

```yaml
alias: "WP MQTT: Set force_update on boiler_nrgtotal if missing"
description: >-
  This automation adds 'force_update: true' to the discovery message of
  'boiler_nrgtotal' when the flag is missing or false, preserving all other
  content.
triggers:
  - topic: homeassistant/sensor/wp-ems/boiler_nrgtotal/config
    trigger: mqtt
conditions:
  - condition: template
    value_template: >
      {% set payload = trigger.payload | from_json %} {{ not
      payload.get('force_update', False) }}
    enabled: true
    alias: Only if force_update flag is false or missing entirely
actions:
  - data:
      topic: homeassistant/sensor/wp-ems/boiler_nrgtotal/config
      payload: >
        {% set payload = trigger.payload | from_json %}{{ dict(payload,
        force_update=true) | to_json }}
      retain: false
    action: mqtt.publish
    enabled: true
mode: single
```

mroeckl, 2025-04-20:
Danke, @thoughtgap , deine Lösung funktioniert super. Hab heute die Automation bei mir eingebaut und eine MQTT-Nachricht mit `force_update=false` und mit fehlendem `force_update` publiziert und, wie erwartet, wurde daraufhin von der Automation wieder `force_update` auf `true` gesetzt.
`wp-ems` im Topic-Name hast du in ems-esp definiert, oder? Bei mir lautet der Topic-Name `homeassistant/sensor/ems-esp/boiler_nrgtotal/config`. Wird das im Feld `Base` gesetzt (siehe Screenshot)?

[image]

thoughtgap, 2025-04-20:
Ja genau, ich habe den Hostnamen und das Base-Topic von `ems-esp` in `wp-ems` umbenannt. Ich habe übersehen, dies in den Standard zu ändern bevor ich es hier poste.

closed 2025-04-26 by mroeckl

DennisM84, 2025-05-24:
@thoughtgap Hier mal das was im Trance zu den Variablen steht.

context:
  id: 01JW183ZVN5G95HJZ53D383TMF
  parent_id: 01JW183ZVN5Y64P3MRDT7PXWPY
  user_id: null

Hier ist die user_id leer. Kann es daran liegen? Muss ich hier noch irgendwelche Daten mit geben?

DennisM84, 2025-05-27:
> [@DennisM84](https://github.com/DennisM84) Die Automatisierung schnappt sich die MQTT Discovery Nachricht (via `trigger.payload`), die das EMS-Gateway mit den Metadaten des Sensors versendet, und versucht, diese um das `force_update` Flag zu ergänzen.
> 
> In deinem Fall scheint die Payload nicht ordentlich anzukommen. Du kannst sie dir im Trace der Automation im obersten Schritt, dem Trigger, unter "Geänderte Variablen" anzeigen lassen:
> 
> <img alt="Image" width="858" src="https://private-user-images.githubusercontent.com/3226639/447238584-5a515d1b-9f27-488c-a98b-e1fea9bd1831.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDgzNzQ3NzcsIm5iZiI6MTc0ODM3NDQ3NywicGF0aCI6Ii8zMjI2NjM5LzQ0NzIzODU4NC01YTUxNWQxYi05ZjI3LTQ4OGMtYTk4Yi1lMWZlYTliZDE4MzEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDUyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA1MjdUMTkzNDM3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzU2YWRmODU0MzNlYmM2ZDRmZjRiYzA1ZmZmMWQ2NjI5Mzg3NzNjY2QyMzYxNzE3YjE3ZWU3MmNkZTU5ODViMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.ytr4U9IlqWYZe58pffAoelmK2kJuqF1MeUTv9BcQ6ko">
> Bei mir sieht `trigger.payload` wie folgt aus (Achtung, mein mqtt-discovery Topic heißt `wp-ems` anstelle des `ems-esp` Standards):
> 
>   payload: |-
>     {
>       "uniq_id": "boiler_total_energy",
>       "obj_id": "boiler_total_energy",
>       "name": "Gesamtenergie",
>       "stat_t": "wp-ems/boiler_data",
>       "val_tpl": "{{value_json['nrgtotal'] if value_json['nrgtotal'] is defined else 0}}",
>       "avty": [
>         {
>           "t": "wp-ems/boiler_data",
>           "val_tpl": "{{'online' if value_json['nrgtotal'] is defined else 'offline'}}"
>         }
>       ],
>       "avty_mode": "all",
>       "unit_of_meas": "kWh",
>       "stat_cla": "total_increasing",
>       "dev_cla": "energy",
>       "dev": {
>         "ids": [
>           "wp-ems-boiler"
>         ],
>         "name": "wp-ems Boiler"
>       },
>       "force_update": true
>     }
> Ggf. würde es helfen wenn du zusätzlich deine gesamte Automation als yaml postest.

Okay, ich habe ein Problem mit dem MQTT. Nach Löschen und neu erstellen der Automation funktioniert es. Nach Neustart des Broker auf der Home Assistent Installation geht es auch. Bei jedem weiteren Triigern ist trigger.payload nicht vorhanden und es wird der Error "Fehler: Error rendering data template: JSONDecodeError: Input must be bytes, bytearray, memoryview, or str: line 1 column 1 (char 0)" geworfen. Eine Idee woran es liegen könnte und wie ich es beheben kann?

Danke für die Hilfe 😅😊